Befaco/MIDIThing2 #1: Clock start latency and general instability

voltagecontrolled, 2024-05-05:
Using MIDI Thing V2 with MIDI in from Elektron Digitakt Mk1, Synthstrom Deluge, and OXI One, with an output set to Clock with 16th notes, the following is observed:

- Nearly all the time, the clock start is significantly lagged
- The clock starts fast and then settles into the correct tempo (i.e. starts at 124, slows to 120)
- Result appears to be inconsistent / unpredictable

The practical result is that modules that start when receiving clock (i.e. Pamela's Pro Workout) will be lagged and/or off beat. 

As for workarounds, having the clock run all the time and using a Start/Stop channel seems to work OK, however this requires using up another output. For modules that have Reset and not Run, I've not yet found a way to get tight sync.

xustafu, 2024-05-06:
Thanks a lot for the feedback. We will look into this later today.
The problplem when sending MIDI clock is that it usually takes MIDI thing a few MIDI clicks to measure the speed then start processing CV clock. We might make this tightener, but it will always will be needing at least a couple of clocks to meassure the time.
Our suggestion will be to use your workaround and keep sending the MIDI messages.
We will look for possible solutions to improve the situation!

Not sure if you are the same person posting in MW. Over there some mention unstabilities in the clock. Is this your case? Are we talking about that clock is shifted one step due to starting late? or are we talking about the clock being unrelaible and draging/running fast at random?

Again thansk a bunch for the feedback!

voltagecontrolled, 2024-05-06:
I'm the Modwiggler poster. I don't think the clock rate is fluctuating, it seems pretty stable. However when you have ST/St set to "on", the clock seems to lag when it starts outputting, whereas if ST/St is off it stays constant.

This leads me to wonder if it may be unrelated to MIDI since my device (Digitakt) is continuously sending the clock signal.

xustafu, 2024-05-06:
St/sp option will stop MIDI clock to be generated after a Stop message is received. 
Maybe the problem here is that clock stops being procesed! meaning that it will take some clicks to get the timing again.
Allow me to check this internally, as keeping clock being processed (although not generated in module) will fix this issue.

xustafu, 2024-05-09:
@voltagecontrolled to wich mail could I send you a test firmware?

voltagecontrolled, 2024-05-09:
@xustafu I received your DM on Modwiggler, I will test this out in a few hours after work. Thank you!

voltagecontrolled, 2024-05-10:
I tested the firmware you sent and found that with the clock set to 16th notes and ST/St on, the issue persists and I still do not get good sync to Pamela's Pro Workout. There is still significant startup lag which creates overall latency.

I have a few other clocked modules, but they all use Reset inputs, and I'm not sure how to get a reset trig to fire on MIDI Thing V2. So I was only able to test with Pam's. 

I did notice one other issue. After doing the firmware update procedure, the module freezes and goes into a boot loop when trying to reload my saved state. Even after rolling back to the original firmware, the saved state from prior to the upgrade fails to load. My saved configuration was as follows:

1. Pitch CV (V1, MIDI Ch. 1)
2. Gate (V1, MIDI Ch. 1)
3. Velocity (V1, MIDI Ch. 1)
4. CC2, MIDI Ch. 1, -5/+5v
5. CC3, MIDI Ch. 1, -5/+5v
6. CC4, MIDI Ch. 1, -5/+5v
7. Pitch CV (V2, MIDI Ch. 2)
8. Gate (V2, MIDI Ch. 2)
9. Velocity (V2, MIDI Ch. 2)
10. CC2, MIDI Ch. 2, -5/+5v
11. CC3, MIDI Ch. 2, -5/+5v
12. Clock, 16th notes, ST/St off

Since the module loads the saved config at boot, this made it seem like the module was dead after firmware update until I realized pressing and holding the encoder for a while causes the module to load a default state, and now it works OK. But any time I try to load that preset, it hangs.

xustafu, 2024-05-10:
So sorry!! forgot to mention that with new firmware, saved statuses will not work, so you need to wipe your saves!
If you want to keep them, go back to previous version, and store them in the computer using our web tool.
But anyway I think that redoing your config should not be too difficult.

> I tested the firmware you sent and found that with the clock set to 16th notes and ST/St on, the issue persists and I still do not get good sync to Pamela's Pro Workout. There is still significant startup lag which creates overall latency.
> 

Is this only happening in 16th? the rest of ppqn divisions work well?

xustafu, 2024-05-12:
@voltagecontrolled  can you confirm that now clock is properlly managed when ST/SP option is ON?

Meaning this:

> However when you have ST/St set to "on", the clock seems to lag when it starts outputting, whereas if ST/St is off it stays constant.
> 
> This leads me to wonder if it may be unrelated to MIDI since my device (Digitakt) is continuously sending the clock signal.

voltagecontrolled, 2024-05-14:
@xustafu I did not notice any difference with the firmware you provided. With ST/St on just sending a clock to Pams and nothing else, there's still startup lag.

I reverted back to the stock firmware to try and resolve the crash I was having which turned out to be due to the corrupted preset.

SergioRetamero, 2024-05-14:
Thanks for helping us understand your needs, it`s important for us to implement the changes that most users would require.

The current clock implementation is as follows:
(In the images, yellow line is the output for Start /Stop port and blue is the output from a clock port)
When the first clock message is received, the module will send a clock pulse:
[image]
I understand you should see no delay in this case.

When you press start/stop there are two cases:
* The MIDI setup your are using to send messages, stops sending clock messages until start is pressed again (for instance default config for Arturia besatsetp) for more than two seconds. In this case, the signals after next start message should be the same as above.
* In case the clock continues after the stop message ( see below with the clock option ST/St Off)
[image]
The main clock will keep record of the steps, so when next start is received, it will send the next pulse in sync with the step:
[image]
As you can appreciate in the image, the start message and clock pulse in this case are not in sync because the clock continued keeping account of the clock steps and sends pulses in the same sequence.
@voltagecontrolled  Is the delay you perceive happening in this situation?

The reason for this strategy is because you may have several clock outputs that need to be in sync and if you keep sending clock messages there might be other hardware in the MIDI setup using the same clock sequence and hence the importance of keeping the pulse sequence coherent in this case.
If I understand your situation, would you like that when the ST/St option in the clock output is set to On, the clock should be reset? This might imply some risk because you will be potentially using a different clock sequence in other hardware.
In this case there will be still a small delay between the start message reception and the next clock message received unless the MIDI sending software is already implementing this synchronisation when sending the start message.

Please, let us know if this explanation helps and how you would like the firmware to perform.

voltagecontrolled, 2024-05-15:
> @voltagecontrolled Is the delay you perceive happening in this situation?

No, I'm experiencing an issue when ST/St is ON. I will try to capture an audio file to illustrate this later, but I will need to re-load the firmware you sent me because I've gone back to stock. 

There is a setting on Pamela's Pro Workout where just a clock signal is sufficient to stop/start the module. So I first attempted to use a single output (#12) on MIDI Thing V2 to drive Pam's. However the clock starts late and fast, then gradually settles into a steady rate but doesn't match the incoming MIDI clock. So instead I have used two cables:

This WORKS:
#11: Clock mode with ST/St off set to 16th notes > PPW Clk input (Set to 4 ppqn)
#12 ST/St mode set to "Start/Stop" (gate high when running) > PPW Run input (set to "Run")

This DOESN'T WORK:
#12: Clock mode with ST/St on set to 16th notes > PPW Clk input (Set to 4 ppqn, RUN set to "CV")

The downside of the approach that works, is I lose a CV input on Pam's, and lose an output on MIDI Thing 2.

janne808, 2024-05-16:
I can confirm that the Midi clock output doesn't line up properly to the actual Midi clock messages. This makes it totally useless in a modular setup when driving a step sequencer for example.

xustafu, 2024-05-21:
This should be fixed now. Pleae try Pre-release binary here:
https://github.com/Befaco/MIDIThing2/releases/tag/Pre_Release_1_1

@voltagecontrolled 
Now, when St/sp option is enabled, clock is reseted, so no shifting should happen. Be warned that if several clocks are configured, not setting the option to ON on some of them might get these running clocks be out of  phase with resetted ones.

If you could provide feedback about this one, would be really appreciated!

xustafu, 2024-05-21:
> I tested this new firmware and the clock is still totally broken just like on 1.0?
Could you be a little bit more specific? what exactly do you mean with totally broken? Are you experiencing the delay on the start of the clock? or it does not work at all?
what setting are you using, modules connected to etc?
Thanks!

janne808, 2024-05-21:
> Could you be a little bit more specific? what exactly do you mean with totally broken? Are you experiencing the delay on the start of the clock? or it does not work at all? what setting are you [...]

My testbench: Midi out from Pamela's New Workout to Midi In on Midi Thing V2, 16th step clock output from PNW and Midi Thing V2 to a mixer. When I start and stop on PNW I can hear a random phase shift on the clock from Midi Thing V2 compared to PNW output.

This is not how Midi clock works, at all.

xustafu, 2024-05-21:
I got a Pamela today. Will recreate your setup and report back!

xustafu, 2024-05-22:
Made a few tests with following code:
[MidiThing2_clock.zip](https://github.com/Befaco/MIDIThing2/files/15402119/MidiThing2_clock.zip)

- With Pamelas situation is stable but not ideal. No crazy jumps not changing tempos as reported. But  clock is lagging a more or less stable number of milliseconds after the first beat.
- Quick test with Octatrack shows same behaviour like pamela. First clock beat is in sync but after the second there is a constant delay. 
- Keystep 37 (using sync out and MIDI out to MIDI thing to compare) is working almost perfect. Clock starting with start message and not getting out  of phase. Only can see around 1 ms. delay that can be due to usual MIDI latency

We need to investigate further internal clock handling of differnet machines to understand why some work fine and other drag behing in phase.

janne808, 2024-05-22:
> * With Pamelas situation is stable but not ideal. No crazy jumps not changing tempos as reported. But  clock is lagging a more or less stable number of milliseconds after the first beat.

I never reported that the tempo was changing, the clock was always stable in frequency but there's a random clock skew that persists during the run yet changes every time you restart the source Midi clock. Essentially the same what you report here.

I tested this with Pamela and Syntakt as the source clock to same effect.

xustafu, 2024-05-22:
Issue should be fixed now!
https://github.com/Befaco/MIDIThing2/releases/tag/Pre_Release_1_1

Configured a clock and set st/sp to OFF. Tested with pamela ok

xustafu, 2024-05-23:
> 
> Installed this new firmware and the issue with clock skew seems to be fixed but I have to turn ST/SP to ON instead of OFF?
> 
This was a Typo/mental slip. Will edit.
Closing issue as solved.

closed 2024-05-23 by xustafu