Befaco/MIDIThing2 #2: CV range not saved

WulliMV, 2024-05-07:
If I set a voice note to a 0-8V range, it resets to 0-10V after some interaction. When I change the range, the Voice does not react anymore. When I save the Preset in global menu and reboot, The setting is back to 0-10V.

xustafu, 2024-05-08:
Managed to replciate it. Issue marked for next firmware update

WulliMV, 2024-05-12:
Hi there - Tried some more to work with the module. It is simply not usable at this stage (or maybe I really have a faulty unit in addition to the firmware bugs):

- Note Glide is not saved as well, the ADSR constantly resets to default values, 
- ADSR & Gate "Continue/Retrig/No Retrig) settings reset to default. 

As soon as I save these settings, after the next power cycle it either resets to defaults or shows the "ants screen" and I have to wipe memory and start again. 

When will the worst of these issues be solved approximately?

Thank you & cheers, Chris

xustafu, 2024-05-12:
We should have a testing firmware will all fixes, for bugs reproduced and identified, ready in a couple of day maximum.

>     * Note Glide is not saved as well, the ADSR constantly resets to default values,
>

Cannot replicate. Tried saving Note Glide and it did saved and loaded right.
Can you specify wich ADSR values are being reset? also, what steps did you took so we can reproduce here?
Saving normally in my unit!
 
>     * ADSR & Gate "Continue/Retrig/No Retrig) settings reset to default.

Also working fine at this unit!

> As soon as I save these settings, after the next power cycle it either resets to defaults or shows the "ants screen" and I have to wipe memory and start again.
> 

Weird indeed. Try wiping saves, then upload the code again.
I do hope the new firm coming will fix youre saving issues!

WulliMV, 2024-05-17:
I tried wiping saves and re-upload the code, it did not work. Should Midi Thing 2 keep the latest settings after a power cycle?

The following behaviour remains consistent:

1. I load my custom patch (see attached json)
2. Readjust all lost parameters (Voltage range, ADSR Settings)
3. Move the gate to an empty port, as the gate where it was before does not work any longer (screen shows triggered gate, but no signal at the port)

Then I have two options:
a) I do a "save mem" on the device, which will lead to a noise screen at the next reboot --> need to reboot with encoder pushed which will wipe again the settings, If I then try to "load mem", Midi Thing 2 crashes again.
b) I do not "save mem" and after the next power cycle I am back to the 4 voice default settings. 

Whichever path, I need to do the complete Setup anytime I turn off power, which is frustrating. 

The question remains: Faulty unit or firmware issue? :-)

[MTConfig2024_May_17_8_21_41.json](https://github.com/Befaco/MIDIThing2/files/15345310/MTConfig2024_May_17_8_21_41.json)

WulliMV, 2024-05-21:
> This should be fixed now. Pleae try Pre-release binary here: https://github.com/Befaco/MIDIThing2/releases/tag/Pre_Release_1_1
> 
> If you co¡uld provide feedback about this one, would be really appreciated!

Hi xustafu

Not sure if it is me as a non-SW engineer: In the manual it says to clone the whole repository and not just download the hex file. But if I do this, I only get this structure, without a hex file. 
Any way I am trying (also just downloading the HEX file and trying to load this onto teensy), the Splash screen still reads 1.0 and behaves the same as before. Any advice on this?

[image]

xustafu, 2024-05-21:
oh... the hex states v1.0!!

that is wrong :( But the file is right. Go ahead wit hthat one and test

WulliMV, 2024-05-21:
So if I download only the *.hex file and transfer this with the teensy loader. I should be good? Then I unfortunately have not yet experienced any improvements. Still nothing can be saved, ADSR resets to default values, random pixel screen when rebooting, needing to wipe all memory and start from scratch.

xustafu, 2024-05-22:
There must be something fishy here...... Can you try with this one?
I just uploaded it to a unit and save was working fine 
[MidiThing2.zip](https://github.com/Befaco/MIDIThing2/files/15399680/MidiThing2.zip)

WulliMV, 2024-05-22:
OK, seems like I need to return the unit. Simply cannot get it to run properly. again random pixel screen after saving, adsr's resetting to defaults, gate stops working when I change note to 0-8V. Seems then really to be a faulty unit... Sorry for spamming your "issue" list with problems seemingly related to a faulty unit.. Best regards, Christian

xustafu, 2024-05-22:
No problem at all!! Happy to try to help and really thankfull for your patience and enthusiasm ^_ ^
So sorry you are taking the module back!! Please tell people in the shop to mention you and this github issue when they open the RMA.

closed 2024-05-22 by xustafu