Befaco/MIDIThing2 #3: Module crash into an unrecoverable state

mneme, 2024-05-08:
Using the web editor seems to have left the module in a non-recoverable state. The screen shows "random static" and toggles on and off.

mneme, 2024-05-08:
[image]

mneme, 2024-05-08:
Well - I spoke to soon. Turns out it's recoverable by holding the encoder down as it started up and then clearing all saved configs.

mneme, 2024-05-08:
Well - I spoke to soon. Turns out it's recoverable by holding the encoder down as it started up and then clearing all saved configs.

Might be something to add to the documentation though.

closed 2024-05-08 by mneme

xustafu, 2024-05-09:
> Well - I spoke to soon. Turns out it's recoverable by holding the encoder down as it started up and then clearing all saved configs.
> 
> Might be something to add to the documentation though.

There is a mention in the manual, but its true it might be good to give  this one some relevance.

WE have a couple of open issues regarding memory save. Like https://github.com/Befaco/MIDIThing2/issues/2
This might be related, as the freeze seems to be caused by corrupted memory.
Stay tunned for new firmware release in the coming days!!