Befaco/MIDIThing2 #4: Save Mem / Load Mem / Pre-Def

WulliMV, 2024-05-09:
I accidently clicked "Pre-Def" in the global menu. This gives various options to overwrite the settings with a template. There is no way to exit this menu (e.g. "Back" as additional option in the submenu). I thought - OK, no worries I saved my settings anyways. So I clicked one of the options in "Pre-Def" and went to "Load Mem". This unfortunately crashes the unit. Seems "Load Mem" does not yet work, so my settings are gone.

xustafu, 2024-05-09:
Created an internal  feature request to implement a back option in the list of pre-def.
Good catch, thanks!!

Tried to replicate crash in Load mem, but could not make it crash. If you can provide some more precise steps to replicate it, then I can forward this to development.

WulliMV, 2024-05-09:
Hm should not have tried.. I thought it might be connected to me trying to load the same memory again instead of another.. it got ung up instead of crashing. Powercycle only brings noise to the screen (see attached image).
[image]

xustafu, 2024-05-09:
Power the module with encoder ON to get out of that screen. Then wipe memory.

Can you please specify the steps you took to get to this screen?

WulliMV, 2024-05-09:
Phew here we are again, the device is back :-). Maybe something was corrupted in memory. I was in memory slot 1 and tried to load memory slot 1 again after something got changed invloluntarily. Then it froze. Now after the memory wipe and Teensy reflash, this does not happen anymore.

xustafu, 2024-05-09:
Not able to replicate the freeze. Did loads of different changes andno luck.
Ther emust be some configthat is corrupting thememory. As this hapenedto other users before.
Lets hold it to next firmware update as save and load will have some bug fixes, so we can keep on testing!
Hopefully we will have something to test in the coming days.

xustafu, 2024-05-21:
This should be fixed now. Pleae try Pre-release binary here:
https://github.com/Befaco/MIDIThing2/releases/tag/Pre_Release_1_1
We have identified some funky bussiness in the memory allocation, so now memory corruptions should be under control.

If you co¡uld provide feedback about this one, would be really appreciated!

jrood, 2024-05-24:
I’ve flashed with the pre-release 1.1 but am still seeing a memory allocation issue, specifically if I create a new “note+adsr” voice and set the adsr shape to linear, and then save.

jrbellido, 2024-05-24:
Hi, I flashed the binary Pre 1.1 and I keep having the same memory corruption issue. These are the steps I follow to reproduce:
- Power on
- Wipe saves -> Y
- Set ADSR on port 3 for voice V1
- Set Note on port 1 (V1) to -5/5V
- Save preset in slot 01
- Power off and on
- Adjust ADSR shape (set attack to 1)
- Save preset in slot 01 (want to overwrite it)
- Power off and on
- Screen appears with random white dots and frozen
- Power off and on with encoder pushed
- The preset (01) is gone and trying to load it freezes the unit

Hope you can reproduce it. Thanks!

xustafu, 2024-05-25:
>
> Hope you can reproduce it. Thanks!
>
Unfortunately could not reproduce it!
Please load the following code following these instructions:

1. Wipe saves in your module.
2. Upload the code. Make sure when it boots spalsh screen states Pre2 v1.1
3. Wipe saves.
4. Reproduce your steps and check if it helps!

[MidiThing2_pre2_1_1.zip](https://github.com/Befaco/MIDIThing2/files/15444288/MidiThing2_pre2_1_1.zip)

If it does not work, contact befaco support to try to replace your teensy board and check it that helps.

jrood, 2024-05-25:
I had initially tried `Pre1 v1.1`. I'm still seeing memory corruption with `Pre2 v1.1`.

xustafu, 2024-05-25:
Please contact support@befaco.org and lets try to swap those units.
In the meantime we will keep investigating

jrood, 2024-05-25:
Thank you. I’ve emailed support. I should also mention that sometimes the memory issue doesn’t result in random dots, but rather a screen that appears functional but is frozen after reboot.

xustafu, 2024-05-26:
As this issue seems to be happening only to some units, we will swap teensy boards for some that are not showing this behavior. Support peepz will make sure of this.
In the merantime, maybe you guys can do some more tests with us?
This problem is related to corrupted save states. This is why we are wiping saves to restore the module. Could you try to save in differnet slots? Also checking how much memory you are using (lil number under save slot selection).
Also, can you try to achieve the same config but starting from different pre-defs. This will provide very useful information for our investigation!

Thanks!

jrood, 2024-05-27:
In my case, I’m only seeing this issue when saving ADSR where I have modified the shape. It doesn’t seem to matter which parameter I change within the ADSR shape or which slot I save it to, the effect is always that after I power off, when I turn back on it is frozen on the last screen I left it on.

xustafu, 2024-05-27:
> In my case, I’m only seeing this issue when saving ADSR where I have modified the shape. It doesn’t seem to matter which parameter I change within the ADSR shape or which slot I save it to, the [...]
Managed to reproduce it!!! will check internally and report back!
Thanks for the feedback!!!

xustafu, 2024-05-29:
New testing firmware!
[MidiThing2_pre3_1_1.zip](https://github.com/Befaco/MIDIThing2/files/15489087/MidiThing2_pre3_1_1.zip)

jrbellido, 2024-05-29:
Hi, I have been testing this pre3 1.1 and seems to be working now. No memory corruption when changing the ADSR shape or playing with presets (save, load, reload,...). I've only been able to spend a little time, but it seems that the unit is already usable for my needs.

Thanks for the support and quick response!

jrood, 2024-05-30:
Confirmed, pre3 1.1 solves the issue for me also! 🚀

xustafu, 2024-05-30:
https://github.com/Befaco/MIDIThing2/releases/tag/PreRelease_3_v1_1

closed 2024-05-30 by xustafu