Befaco/MIDIThing2 #5: "Voice 1" assigned port suddenly "Voice 2"

WulliMV, 2024-05-09:
When changing Settings on some port allocated to Voice 1, I experienced e.g. an ADSR assigned to Voice 1 as well suddenly is assigned to Voice 2.

xustafu, 2024-05-09:
I have seen this happening with Gate option, and we have an issue open for it.
Can you provide steps on how to replicate this on ADSR? Just tried with several combinations but did not managed to make it change.
Were any save involved during the process? I got the change of channel after save only!

WulliMV, 2024-05-09:
Hi - yes, change of channel happened as well. I think it may have happened with another option. or when setting up another port with a feature, that cannot occur twice.. I'll try if my midi thing starts working again ;-)

xustafu, 2024-05-14:
Copy paste from MW:

Still having an issue with a gate voice not saving correctly. I set the module to 4 voices with voice 3 and 4 having gate outputs. Next I save the config and shut off the system. When powered back on the midi thing v2 allocated the gate for voice 3 to voice 4 and the gate for voice 4 to voice 5. I tried reloading the saved config and nothing changed. When I fixed the gate for voice 3, it also fixes for voice 4. Not really sure what I am doing wrong but at least a workaround fix was easy to find!

xustafu, 2024-05-21:
Not sure if this will be solved by new firmware, as I cannot reproduce it.
But its defiently worth a try! Please give it a go and report back! Thanks!!

xustafu, 2024-05-30:
https://github.com/Befaco/MIDIThing2/releases/tag/PreRelease_3_v1_1

closed 2024-05-30 by xustafu