Befaco/MIDIThing2 #6: Wish: Multiple ports with same function for the same voice

WulliMV, 2024-05-09:
I'd like to have from voice 1 each a gate and a pulse (trig). This is currently not possible. I cannot assign a second "gate" to voice 1. I say "gate" because I need to use a gate function with the option set to "pulse" for a trig.

xustafu, 2024-05-09:
This have been requested by our beta testers. For example to get several ADSRs from one note etc...
This might be achived with a "kind of Unison" mode, where a note is assigned to several voices.

Will leave this issue open as a follow up for this feature. Hopefully we can get it implemented after the first wave of bug fixing and publishing the code!

xustafu, 2024-05-22:
Merging features requests in one:
https://github.com/Befaco/MIDIThing2/issues/13

digitalies, 2024-07-25:
Crazy ask but seems appropriate to add on here. But would it be possible to implement midi for each envelope's parameter that has the same voice? for example? If 2 envelopes were on the same channel, they would end up sharing the same midi cc's for their parameters.

Some midi controllers are limited to one channel of cc's at a time. It would be nice to be able to remap the "Things" default midi cc's to be able to have separate control on a selected parameter. 

I understand that would mean a whole different approach to the underlying system. No idea how or if it could be changed. Ideally though, every function would be mappable, lfo shapes too please.

Perhaps a matrix the users can access? Is there enough space in the things memory for such nonsensical programming? 

I'd pay for another thing or two just for the implementation. :)

xustafu, 2024-07-25:
> Crazy ask but seems appropriate to add on here. But would it be possible to implement midi for each envelope's parameter that has the same voice? for example? If 2 envelopes were on the same [...]

Let me open another featuer request.
Mapping parameters to differnet midi messages was considered but not implemented. Cannot enure when this feature might be implemented but its defiently worth a festure request!

jjobse, 2024-11-20:
+1 for the original request for multiple ports with the same functionality. Is this feature on the dev teams roadmap? 

Here are some examples of where I need it regularly. 
One example I am using a E370 which is 4 oscillators. Every time I want to switch between a poly patch and a unison or duophonic patch it requires rewiring my entire rack and the use of buffered mults when it should just be a quick config change on the midi thing v2. This is typically a basic option in all midi to cv modules. It would be very helpful to have these available on the midi thing v2.

In another example not using gate/note cv, I have a monophonic patch configured and would like to have the gate trigger multiple synced lfos at different speeds for modulating the various parameters on the E370. When using the midi thing v2 this is not possible, I currently have to use a mult to pass the gate to an acid rain maestro. Otherwise I just use my Zoia to do this all in one module.

xustafu, 2024-12-02:
Closing this issue as we will have a feature request thread for now.
https://github.com/Befaco/MIDIThing2/issues/46

closed 2024-12-02 by xustafu