Befaco/MIDIThing2 #7: Wish: "Load Mem" through a "global" Midi Thing midi channel

WulliMV, 2024-05-09:
When the "Load Mem" will work properly, It would be great to be able to change Memory Slots with a PC Message. For this, Midi Thing would require to have a global Midi Channel set.

xustafu, 2024-05-09:
Maybe an option in Program change message that will go thru load statuses? or even pre-defs?

WulliMV, 2024-05-09:
Yes kind of, but different. following scenario: I could have a memory, where the filter is set to 50% closed, with a slow evolving adsr, squarewave shape ona an oscillator and a lot of reverb mixed in. Each controlled via a constant CV value out of midi thing 2 into corresponding CV in's on the modules. In another memory, I have a snappy envelope, filter fully open, oscillator shape changed to saw, no reverb. Again set through multiple constant CV Values out of midi thing 2. Then I could change these by sending a PC message to midi thing to change memory. not just the constant CV value of one port.. does this make sense?

xustafu, 2024-12-02:
Closing this issue as we will have a feature request thread for now.
https://github.com/Befaco/MIDIThing2/issues/46

closed 2024-12-02 by xustafu