Befaco/MIDIThing2 #8: Note Priority: "Last" not working properly

WulliMV, 2024-05-09:
Note priority only works with 2 pressed keys. If more than 2 keys are pressed, it reverts to the first pressed note.

xustafu, 2024-05-09:
Tested with pre-def 1, sending 4 notes that are assigned to each channel. Working correctly.
Pleaseprovide precise stepstoreplicate your issue.
According to user manual, module is working as described. "Last" will give priority to the last note receives, meaning the  new note will steal the oldest note.

closed 2024-05-09 by xustafu

WulliMV, 2024-05-09:
Works now flawlessly - memory reset helped...

xustafu, 2024-05-09:
Nice bit of information! might be useful to know that memory corruption might go this deep. 
Will keep up trying to replicate!!!

WulliMV, 2024-05-09:
oh, still something: if I press keys, e.g. four in a row and start to release again one after the other, it jumps back to the first instead of the last. here it is again ;-D

xustafu, 2024-05-09:
I load Pre def-1. Priority is selected to last.
Pressing four keys in a MIDI keyboard, having the four voices selected. Then I release one by one. It acts normally.
Cannot replicate. Please elaborate steps to get the error

WulliMV, 2024-05-10:
ok I see. the issue I describe is in monophonic mode, so only one voice with "last" set as priority. then the behaviour i'd expect after releasing the fourth key again that it goes back to the third, not the first note