Befaco/MIDIThing2 #9: Monophonic Note Priority

WulliMV, 2024-05-11:
ok I see. the issue I describe is in monophonic mode, so only one voice with "last" set as priority. then the behaviour i'd expect after releasing the fourth key again that it goes back to the third, not the first note

_Originally posted by @WulliMV in https://github.com/Befaco/MIDIThing2/issues/8#issuecomment-2104108452_
[image]

            
I put a midi track together to illustrate.. In this keyorder I'd expect the green path, but currently the red path happens.. IF I am in a patch where I only have one voice, monophonic, proirity set to "last". Tried to attach my *.json but I cannot.

WulliMV, 2024-05-11:
[image]
[MTConfig2024_May_9_17_44_21.json](https://github.com/Befaco/MIDIThing2/files/15281934/MTConfig2024_May_9_17_44_21.json)

xustafu, 2024-05-12:
WE have duplictaed this. Seems that algorythm only applies to "stealing" of notes, we need to implement the algorythm to "recovering" notes.

So setting this to feature request and we will work on it!!

closed 2024-12-02 by xustafu