Befaco/MIDIThing2 #15: Changing note voltage range disables voice's gate output on v1.1

voltagecontrolled, 2024-05-26:
Recently updated to V1.1 and noticed that if I change the voltage range of a voice note output, I stop getting output out of that voice's gate outputs until I reboot the module, or change the output. 

In my example, I have voice 2 set to MIDI channel 3 with pitch on 7 and gate on 8. Modifying the pitch voltage on 7 causes the gate on 8 to stop working until I reboot the module. The gate output flashes on the screen as though it's generating output, but nothing comes out of the output. 

If you modify the type of gate output (even setting it back to the same voice's gate) or save and reload the preset the gate will start working again. Though as previously noted, modifying the voice causes the pitch voltage to go back to default 0/10v.

_Note: I have plans to return the module to my retailer on Tuesday, so regrettably I won't be able to help with testing._

xustafu, 2024-05-26:
Thanks a lot for the feedback! And sorry to hear you will be returning the module.
We are still i npre- release of 1.1 so this will be addressed for the officilal version.
Thanks a lot for the help and patience!

xustafu, 2024-05-30:
Fixed here
https://github.com/Befaco/MIDIThing2/releases/tag/PreRelease_3_v1_1

closed 2024-05-30 by xustafu