Befaco/MIDIThing2 #16: MIDI adapter for  Moog DFAM (MAFD or the "THING")

RomeoFF, 2024-06-28:
Would it be possible to add the possibility to translate midi to control the Moog DFAM (and potentially other basic sequencer). This means translating a MIDI note information into up to 7 pulses, in order to "trig" a specific event in a loop (and play like that as an instrument rather than a sequencer). There is a 2hp module done just for that (MAFD from sonoclast) but it would be much more convenient to have this possibility directly in the MIDI thing…

xustafu, 2024-06-28:
Are we talking about something like this?

https://github.com/dimant/FlexMAFD

RomeoFF, 2024-06-28:
That's exactly the idea ! Thanks for the share, I didn't notice that there was already an open source code for a teensy. It should be possible to adapt the flexMAFD trigger output for the MIDIthing, right ? (here is a video that explain well how to use it : https://www.youtube.com/watch?v=VF6GRCjxO0Y)

xustafu, 2024-12-02:
Unfortunately these changes are out of scope for now. I suggest to keep track of the repo and once code is available you can try to implement this yourself!

closed 2024-12-02 by xustafu