Befaco/MIDIThing2 #26: Select Bus Data Transmission

Genfin, 2024-08-19:
Hello ! 

Could the Midi Thing V2 transmit data through the Select Bus pin? This could be particularly useful for recalling presets stored via the Select Bus. 
If Midi Thing V2 had a Select Bus mode, it could convert incoming Program Change (PC) messages from any MIDI channel into Select Bus commands. For example:

 PC1 -> Midi Thing -> Select Bus Preset 1
 PC2 -> Midi Thing -> Select Bus Preset 2
 ...and so on.
 Would this feature be possible to implement?

xustafu, 2024-09-30:
What Bus do you mean? The one in the power cable?

Gen-finesse, 2024-09-30:
Yes i'm talking about the bus board. Sometime when the last two pin are not used, you can send midi data through those pin . It's called select bus protocol. (Maleko, Shakmat, Expert sleeper use it)

Gen-finesse, 2024-09-30:
But i have to correct my precendent post , it has to be on midi channel 1 (select bus only respond to this channel)

xustafu, 2024-10-03:
Unfortunately those pins are unconnected in the hardware!

So this feature is not feasible without modifying your hardware.
Closing the issue as out of scope.

closed 2024-10-03 by xustafu