Befaco/MIDIThing2 #29: Source

mostl33, 2024-09-18:
Heya guys,

any updates on a relase of the sources? Would love to tinker and maybe teach my MidiThing2 some new tricks :).

Thanks for a greate module!

~ML~

xustafu, 2024-09-30:
Soon!

You can always contact @SergioRetamero  if yo ucan take a look to code before release!

mangobrain, 2025-07-30:
10 months later, any update on this?

xustafu, 2025-07-31:
@mangobrain we are mega close!

I will invite you to the private repo, so you can take a look and help us tidy up so we can publish faster!

mangobrain, 2025-08-03:
Thanks! I saw the invite - I don't have a ton of free time at the moment but will see if I can get it building next time I have a free evening or two.

nikodemus, 2026-01-30:
@SergioRetamero, I'm interested as well. I'd like to take a looking at how tricky the following would be:

- enforcing a minimum gate-down period between noteOff/noteOn
- sample & hold: ie. changing an output's value based on a noteOn on another channel, to allow combining notes from one sequence and gates from another without an additional S&H module, etc
- per-channel filtering options for MIDI through

Time is hard to come by, but I'm going to be an optimist this early in the year...

nikodemus, 2026-10-20:
>     invited!! :)
@xustafu I missed the invite during the summer and it expired (not that I had time then...), but I'm again hoping to find time either over Xmas or early in 27. Can I bother you for a repeat invite?

xustafu, 2026-11-04:
> 13\. > ```
>          > invited!! :)
>          > ```
>          
>          
>          [@xustafu](https://github.com/xustafu) I missed the invite during the summer and it expired (not that I had time then...), but I'm again hoping to find time either over Xmas or early in 27. Can I bother you for a repeat invite?

re-invited you!
We are very close to finish midithingy code, so you are in time for the party :)