Befaco/MIDIThing2 #30: Bug report note range

jotel-california, 2024-09-23:
If you add or remove a module to a voice, all note range boundaries previously set are deleted. For example, I have voice 1 with CV/Gate. Cv is set up so C0 is my 0V point.
Note range is 48-127 for example.
If I now add for example an LFO or Adsr to voice 1, the note range previously set is now gone and back to 0-127. 

annoying, because I always need to remember what my ranges were, before I add sth new.

xustafu, 2024-09-30:
Hi! We are currently working on firmware 1.2 so took the chance to fix this one. Please give a try to following firmware. Remember you need to wipe saves before uploading (See upload instructions  [here.](https://github.com/Befaco/MIDIThing2/releases) )
[MidiThing2_1_2_TEST2_FIX.zip](https://github.com/user-attachments/files/17190288/MidiThing2_1_2_TEST2_FIX.zip)

jotel-california, 2024-10-02:
Hi there, That was quick, thanks!Sadly im on Vacation and will be back in 2 weeks, so I can‘t test it until then.Best, Jo

jotel-california, 2024-10-17:
Hey Xustafu. I just tried the new firmware. The Note Range bug is fixed. However, I found a new bug. When you have midi clock coming in from the midi in and and the usb host, the bpm is added together instead of one taking priority. Meaning if my midi in runs at 120 and the Usb Host at 80bpm, the midi thing will run at 200bpm. Also I was not able to send out midi clock via the usb host ports 2-4. No matter what I check/uncheck, it does not output clock data.Best, Jo

xustafu, 2024-10-17:
Opened an issue for clock bug.
https://github.com/Befaco/MIDIThing2/issues/37
closing this one.

closed 2024-10-17 by xustafu

jotel-california, 2024-10-18:
Short update, I’m not sure if this was the case with earlier firmwares, but I have a lot of hanging gates (notes) when I pause my sequencer.
Maybe the Midi Stop Command can turn all notes off? Otherwise I need to pause my seq at juuust the right time.

Best, Jo

xustafu, 2024-10-20:
lets follow up in the other issue, please. Will uplopad new firmware there