Befaco/MIDIThing2 #33: clock bug?

besb1n, 2024-10-08:
if you switch to internal clock, the start/stop stops working

xustafu, 2024-10-08:
Tried following code and worked for me using a keystep from MIDI Host. Can you please check?
[MidiThing2_1_2_TEST_3.zip](https://github.com/user-attachments/files/17296247/MidiThing2_1_2_TEST_3.zip)

besb1n, 2024-10-08:
i mean when you stop/start the internal clock on the midithing, it does not send start/stop messages. i tried it with your new code, still nothing.

besb1n, 2024-10-31:
just downloaded the new test3 firmware and this is working now. however now i cannot load saves onto the device through the web app.

xustafu, 2024-11-03:
> just downloaded the new test3 firmware and this is working now. however now i cannot load saves onto the device through the web app.

We have a testing version for FW version 1.2. You can test drive it at your own risk :)

https://befaco.org/midithing2test/

Any feedback is welcome!!

besb1n, 2024-11-03:
its working mostly. it seems 1 save file in particular doesnt work.
[main.json](https://github.com/user-attachments/files/17612332/main.json)

besb1n, 2024-11-03:
when you load that save through the browser it will not update on the module. but if you load it through memory it will work fine.

besb1n, 2024-11-04:
start/stop still not working with internal clock. i dunno if it was briefly working before or i just imagined it...

xustafu, 2024-11-04:
> when you load that save through the browser it will not update on the module. but if you load it through memory it will work fine.

Was this save done with v1.1? if so its normal it might not work.

> start/stop still not working with internal clock. i dunno if it was briefly working before or i just imagined it...

True! Reported as bug!

besb1n, 2024-11-06:
yes it was probably done through v1.1

closed 2024-12-02 by xustafu