Befaco/MIDIThing2 #37: Clocks from multiple sources being summed + Host 2-4 not sending clock

xustafu, 2024-10-17:
When you have midi clock coming in from the midi in and and the usb host, the bpm is added together instead of one taking priority. Meaning if my midi in runs at 120 and the Usb Host at 80bpm, the midi thing will run at 200bpm. Also I was not able to send out midi clock via the usb host ports 2-4. No matter what I check/uncheck, it does not output clock data.

xustafu, 2024-10-20:
Please check this firmware. Start stop message is sent and all note off mesage as well.
Check also USB hosts!
[MidiThing2_1_2_TEST_3_FIX.zip](https://github.com/user-attachments/files/17449897/MidiThing2_1_2_TEST_3_FIX.zip)

xustafu, 2024-11-05:
Solved in v1.2 TEST3
https://github.com/Befaco/MIDIThing2/releases/tag/v1.2_TEST3

closed 2024-11-05 by xustafu

modularev, 2024-11-20:
not sure if I overlooked another open issue but on v1.2_TEST3 the Clocks from multiple sources are still being summed? You could either change the menu for the channel selection in the clock related menu for USB + MIDI / USB / MIDI or maybe just change the clock selector between INTERNAL / MIDI / USB?

Is there a timeline when you'll release the sourcecode? happy to take look at it...

xustafu, 2024-11-24:
> not sure if I overlooked another open issue but on v1.2_TEST3 the Clocks from multiple sources are still being summed? You could either change the menu for the channel selection in the clock related [...]

From Test3 we have changed the functionality of clock column in the midi routing screen.
Now the clk colum enables or disables clock handling per source. So to avoid weird clocking issues, disable all clock sources except the one you want to use.

xustafu, 2024-11-24:
@modularev during the afternoon I will send you an invite for our internal repo.
We hope to release the code after official v1.2 firmwqare release, but will be nice to have fresh eyes over it before that :)

xustafu, 2024-11-24:
@modularev sent you an invite,.Most issues are in spanish, sorry. Also keep in mind that code is barely commented and still in need of clean-up!
Ask @SergioRetamero any question you might have or open issues over there!