Befaco/MIDIThing2 #38: Hanging Gates

jotel-california, 2024-10-20:
When you stop any playing midi sequence while a note is being held, the gate out stays high. It would be great, if the Midi Stop Command would set all gates low. Maybe implemented that this can be changed in the settings.

xustafu, 2024-10-21:
Can you try this code?
[MidiThing2_TEST3_All_note_off.zip](https://github.com/user-attachments/files/17458965/MidiThing2_TEST3_All_note_off.zip)

Thanks a lot for the feedback and the patience :)

jotel-california, 2024-10-21:
Perfect, that fixed the problem. Hanging notes are now gone. 

No Problem. I rather have a few bugs and the company cares about the project, rather than a „finished“ product that gets abandoned after launch.

If I have time I help. 

Jo

closed 2024-11-03 by xustafu