Befaco/MIDIThing2 #40: Hanging gate with MPC Live when playing chord with more notes than configured voices

jjobse, 2024-11-01:
Using an MPC live in pad perform mode if I send a chord to a single voice then the gate hangs indefinitely. The only way to get it to stop is to switch to a single note and play that by itself. 

This seems to happen whenever you send a chord with more notes than are configured on the midi thing. For example if I play a 5note chord with 4voices configured it will cause the hang behavior. 

I found I was able to replicate this 100% of the time on the MPC Live on a midi track within the Pad Perform screen.

xustafu, 2024-11-03:
Gate behavior works that it only comes down (or trigers Release stage in ADSR) when a Note OFF arrives. So if chord is infinite and/or no note OFF are sent, gate  will not come down.
Get a MIDI monitor on your computer and check how Note OFF are sent when overlaping notes happen or you send long ones.

jjobse, 2024-11-05:
Hey there, no problem. I have attached my logs, a screenshot, and a video.

- I captured the midi from the MPC on my PC using a ESI MIDIMATE EX cable and put the output into the mpc_chord_midi.csv file.
- I then captured the midithingv2s midi on the device itself using my camera.
- I also captured the gate hang behavior in a short video. You can see the gate go high and then hang there without closing.

For this test I configured the midi thing v2 to have 1 voice and 1 gate attached to that voice. From the MPC I sent a 3-note chord which resulted in a hung gate.

I can see all of the midi messages are being sent as expected. However the midithingv2 is not responding accordingly. Maybe it has to do with the timing of the note off messages being too close to each other (.001 secs apart)?

Please let me know if there is anything else I can do to assist.

[mpc_chord_midi.csv](https://github.com/user-attachments/files/17625230/mpc_chord_midi.csv)

[image]

https://github.com/user-attachments/assets/fb7ac1e7-a61c-40c4-867f-bb428e83579e

xustafu, 2024-11-05:
And this example, sending those three notes from the log, does hang the gate?

Cannot reproduce the issue with a MIDI keyboard!

xustafu, 2024-11-05:
Replicated in bitwig.
With two notes sent at the same time with the same lenghth it is fine, from the third note gate hangs.
Reporting as a bug!

jjobse, 2024-11-16:
After some more testing it looks like round robin mode is broke in this build. Also when playing chords with not enough voices configured the note out jumps around for an unstable sounds.

jjobse, 2024-11-20:
To expand on this a little more it appears when the chord is struck the midi thing is taking some time (fast like a millisecond) to pick which note to play. This is the little blip I am hearing. You can see the note jumping around on the midi thing GUI as well. Maybe this is due to note priority? I will check again to see if note priority has any impact on this behavior.

xustafu, 2024-12-07:
@jjobse Please try this firmware:
[MidiThing2_GATE_FIX.zip](https://github.com/user-attachments/files/18048656/MidiThing2_GATE_FIX.zip)

jjobse, 2024-12-07:
Works great, bug is squashed. Thank you for all the assistance

closed 2025-01-29 by xustafu