Befaco/MIDIThing2 #42: unstable clock

besb1n, 2024-11-07:
test 3 v1.2

getting around .3 bpm drift on incoming clock

xustafu, 2024-11-24:
Difficult to reproduce in our side.

Tested with Bitwig connected via USB device and with a keystep pro via USB Host .
Keystep was 99% of the time stable with some .1 movements. While bitwig was showing this .1 movement a little bit more often.

None of the cases showed a sustancial time isse on prolonged sessions.

besb1n, 2024-11-24:
i tested on ableton/hapax/perkons. all were getting a total range of .3 drift.

jotel-california, 2024-11-26:
If I may chime in, clock works fine on my end. 
I have hooked it up to a multiclock, so I know that my source clock is stable.

xustafu, 2024-12-02:
> i tested on ableton/hapax/perkons. all were getting a total range of .3 drift.

Made a quick test to compare source clocks stability. This might also help you find out the source of the unstability.
Sent clock from Bitwig to MIDI thing, then from MIDI thing via OUT to a beatstep pro.
We could see that clock was juggling around 0.1 bpm at around 70 BPM. Then set the clock up to 200 bpm and saw that the bpm was changing up to .9 ~ 1.1 BPM.

Then connected straight from our audio interface to beatstep, and could see that the unstability was the same. So in our case the source of unstability is Bitwig/our midi interface not MIDI thing.

Please do this same testing on your side as DAW tend to be pretty unstable in general terms.

> If I may chime in, clock works fine on my end. I have hooked it up to a multiclock, so I know that my source clock is stable.

This add up to our tests. A stable source of clock will make MIDI thing behave stable.

We are closing this issue. Please feel free to re open if you find MIDI thing misbehave with stable sources of clock.

closed 2024-12-02 by xustafu

besb1n, 2025-01-20:
yeah i just updated and tried it with a midronome and didnt get any drift. its a problem with both my hapax and perkons unfortunately.

xustafu, 2025-01-20:
> yeah i just updated and tried it with a midronome and didnt get any drift. its a problem with both my hapax and perkons unfortunately.

Thanks for coming back to us!