Befaco/MIDIThing2 #43: Critical Bug Fix: CV Output is not tracking correctly.

jotel-california, 2024-11-14:
Using the newest 1.2 Test 3 Firmware.

I always wondered why my sequences sometimes sound out of tune, and here is my answer. The midi thing does not track correctly. 
If I tune Midi Note zero to a C, I get substantial errors. 
1st oct = -15ct
2nd oct = -20ct
3rd oct -26
4rd oct -28
5th oct -29
6th octave -32ct

I tried 3 other CV sources, and all other sources track correctly, so It's definitly the midi thing
I hope this can be fixed in software.

jotel-california, 2024-11-14:
here is a demonstration with video: https://youtu.be/M3KUzgGkbFM

jjobse, 2024-11-20:
Hello. Have you tried measuring the note voltage straight from the midi thing v2? I had a similar issue but it was my oscillator not tracking properly. FWIW I have seen very stable voltage across multiple octaves from the midi thing v2. Most oscillators have a calibration trim pot on the back you can tweak to resolve this issue. Also once you confirm there is a 1v different in voltage across each octave make sure you let the oscillator warm up for 15mins or more before you start measuring. I used my Zoia to measure the voltage and the pitch ranges with great success.

jotel-california, 2024-11-20:
Hey. Yes I did, and just plugging the midi thing into the multimeter and there the voltages read ok. Although „0V“ actually is 0.2V on the midi thing. Then it scales correctly, 1 oct above is 1.2 and so on. However as soon as you plug it into anything it stops tracking correctly. I can rule out any oscillator flaws 100%.I tested it with 6 different oscillators across 2 systems with different PSUs, analog as well as digital oscs. The behaviour is the same, so it is 100% the midi thing. Are the outputs buffered? Cause if not, the impedance if the cv in could distort tracking. Jo

jjobse, 2024-11-20:
Ah interesting ok I’m tracking. Yea mine is not exactly 0vs as well but does track 1v per octave. Do you see the same behavior if you stick a buffered mult in between the midi thing and the oscillator? One of my older Pittsburgh modular midi and oscillator combos had that same issue and I had to send the midi device in for repair/replacement. 
Sorry to hear about the issues hopefully you can get it sorted out. This is my first befaco product and I am a little concerned with all the bugs I see with it (especially with such a high price) so I have been checking this page constantly for updates. I love the screen, the way you can configure it to work with many different oscillator types, and the number of outputs (I can get poly into the rack and multiple mono voices at the same time). But this is one of the most bug filled eurorack devices I have ever dealt with and it’s missing some pretty basic features. Makes me wish my zoia had an expansion bay.

jotel-california, 2024-11-20:
Yes, a buffered mult does not help. Sure, the extreme voltage drop when passively splitting the cv is gone, but it still does not track exactly.

xustafu, 2024-11-24:
>  But this is one of the most bug filled eurorack devices I have ever dealt with and it’s missing some pretty basic features. 

Sorry for such a dispointing user experiece.
Please feel free to keep on reporting bugs and we will do our best to fix them all. Thansk a lot for your valuable feedback.

jotel-california, 2024-11-26:
> > Using the newest 1.2 Test 3 Firmware.
> > I always wondered why my sequences sometimes sound out of tune, and here is my answer. The midi thing does not track correctly. If I tune Midi Note zero to a C, I get substantial errors. 1st oct = -15ct 2nd oct = -20ct 3rd oct -26 4rd oct -28 5th oct -29 6th octave -32ct
> > I tried 3 other CV sources, and all other sources track correctly, so It's definitly the midi thing I hope this can be fixed in software.
> 
> Tested on a couple of units and could see a total deviation of 10 cents. Which is not perfect but pretty decent. Did you tried with other VCOs? is this 29 cents deviation constant?
> 
> Anyway, allow us a couple of days to cook a calibration procedure so you can test on your side. We will come back with a test firmware for you!

Alright, excited to see what you can come up with!

the tracking error is a little bit different with different vcos. Tried it with 6 different oscs in 2 different systems with different psus. 
digital aswell as analog. 
But it definitly is similar so maybe not -29ct, but -22ct or similar but I guess that could really be the tracking error of the oscillators themselves. 
Or one other thought I had is that if the outputs of the midi thing are unbuffered, the impedance of the cv inputs could affect the output level.

xustafu, 2024-11-26:
@jotel-california  Here you go!
[MidiThing2_calibra.zip](https://github.com/user-attachments/files/17921211/MidiThing2_calibra.zip)

Calibration pocedure is available in the Options menu of the CV option of a voice:
[image]

It is a bit raw UI wise, but functional. We might be adding some lettering to make it clear, but for now it is good to test!
So we will be calibrating two points, so the slope of the conversion will be adjusted. the procedure is:

Connect the CV output in MIDI thing 2 to your Oscillator V/oct. Then connect your oscillator to a Tuner.
 Then navigate to CV Options menu.

- Send MIDI Note 24 (that should correspond to 2v) with your preffered MIDI controller. Go to 2v cent adjustment A. Tweak it until your oscillator is exactly two octaves over the initial point. The numbers will be the amount of cents you are moving the CV up or Down the original value.
- Send MIDI Note 96 now  (that should correspond to 8v) Go to 8v cent adjustment B. Tweak it until your oscillator is exactly eight octaves over the initial point. The numbers will be the amount of cents you are moving the CV up or Down the original value.
[image]

Now this CV out should have improved tracking linearity! Save the configuration in order to keep the changes!

jotel-california, 2024-11-26:
I meant +12ct on the 0V point, sorry.

closed 2025-02-18 by xustafu