Befaco/MIDIThing2 #46: Feature request thread

xustafu, 2024-12-02:
Feature request thread. We will gather all requests here for repo tidiness.
We cannot ensure when a feature will be implenmented or if it will. But lets keep track here.

-

mangobrain, 2025-05-30:
It would be nice to have an ADSR mode which only retriggers during the release stage. The current behaviour makes playing monophonic voices with long release times very awkward: either retrigger is on, and every note always restarts the envelope from the beginning, even when playing legato; or retrigger is off, legato notes behave as expected, but as soon as the last key is released, you have to wait for the entire release stage to finish before the envelope can be retriggered.

Typically an ADSR *would* retrigger during the release stage; and rather than always strictly retriggering from the start, the attack phase would begin part way through, picking up from whatever voltage the release had dropped to when the new note on arrived. As a reference, patching up an ADSR on Mutable Instruments Stages & compatible clones, using a ramp for attack, ramp for decay, looping hold segment for sustain, and ramp for release (exactly as per the ADSR example in its quick start guide) behaves this way: if the gate goes high again during the release ramp, it moves to the attack ramp, without dropping the voltage to zero.

jotel-california, 2025-06-13:
Cool thing would be adding the note ranges (e.g. midi note 24-127) to the gate/velocity page aswell.
This way you could trigger auxiliary gates from one midi clip. 
For example you could use midi notes 0 and 1 to trigger certain things, but leave the rest of the notes no play the melody.
Right now the only way to achieve this is to use a different midi track in your daw, or sacrifice a channel on the midi thing for another CV out, since you can only set the note range in the CV options page.

laddernoise, 2025-06-20:
Hi!
Thanks for your amazing module!
It could be so great to implement a glide (portamento) option that only activates when incoming MIDI notes are overlapping. This would emulate classic legato behavior, which is widely used in expressive monophonic playing.

Thank you so much!

Kevin

hoprocker, 2025-08-03:
> Hi! Thanks for your amazing module! It could be so great to implement a glide (portamento) option that only activates when incoming MIDI notes are overlapping. This would emulate classic legato behavior, which is widely used in expressive monophonic playing.
> 
> Thank you so much!
> 
> Kevin

This would be great. It'd make it a lot easier to play monophonically (without having to worry about releasing the previous note first).

xustafu, 2025-08-13:
> This would be great. It'd make it a lot easier to play monophonically (without having to worry about releasing the previous note first).

@hoprocker can you try this code? Now one of the Glide options is overlap. Is this what you were looking for?

[MidiThing2_1_2_2_TEST4.zip](https://github.com/user-attachments/files/21760475/MidiThing2_1_2_2_TEST4.zip)

hoprocker, 2025-08-14:
While we're at it, I'd also like to +1 the idea from #6 , _especially_ for ADSR envelopes. This could emulate the classic VCO/VCF envelope architecture, but could also mean VC?, depending on what you route it to (wavefolder? timbre? noise mix on your VCA??). 

This can be done right now by sending gate/trigger to an external envelope generator with input normalling, but I can definitely picture situations where I would have 1 note + 5 ADSR going to various CV sources to create some sort of swarm effect (everything explodes on the A part and resolves on the DSR).

Thank you for this module! It's fun to see the Teensy showing up in something other than O_c.

digitalies, 2025-12-09:
Could you scale the CV output for Buchla's 1.2 volts per octave?

hoprocker, 2025-12-17:
> Hi! I'm back, so sorry i dind't make some modular synthesis since a while... with the TEST 6, I didn't success to make glide at all... even in lineal mode, like if everything was disable.. I see all the mode avalaible like overlap etc but not working at all.. i don't know what i'm doing wrong now but it's very strange..

I just went through the last few firmware revisions, and here is what I can confirm:

1.2.2 TEST4: glide works
1.2.2 TEST5: glide works
1.2.2 TEST6: glide DOES NOT work (also this version says TEST5 on startup, so I triple checked I had the correct one loaded)
1.3 RC1: glide DOES NOT work

Hth!

Faderline, 2026-10-21:
Hi guys,

simple version: add 'scale' option for midi CC to CV.

I use an octatrack to send CC values and LFOs to operate Control Voltage around my modular. They are variously 0v-5v, 0v-8v, -5v-5v, 0-10v. in anything other than a 10v range, I have to remember per destination if my CC range stops at 127 (10v), 64 (5v), or 102 (8v).

It's a pain even whilst sitting in the studio, but forgetting which is which live and running a CC that maxes out the voltage for a chunk of the LFO cycle is a real problem. You've got the clamp for voltage, and the offset for -5v-5v in place, I am sure scaling is such a lightweight add and would make the clarity/ease of use at the Octatrack so much greater.

Does anyone know how we find out if features will be added or when, following a request?

Thanks everyone.

@xustafu

bartofzo, 2026-10-25:
Hi,

feature request: “length” mode for the LFO (and clock) multiplier, for polymetric patterns

Currently, the multiplier speeds up the selected division: 1/8 × 3 means three cycles per eighth note. That’s great for polyrhythms, but every multiplier other than a power of two puts the LFO’s cycle off the beat grid.

It would be very useful to have an alternative multiplier mode where the value sets the length of a cycle in units of the selected division. For example, with 1/8 selected:

×1 = a cycle of one 8th note
×2 = a cycle of two 8th notes (a quarter note)
×3 = a cycle of three 8th notes (a dotted quarter)
×5 = a cycle of five 8th notes, and so on

This would make polymetric patterns possible: LFOs with cycle lengths of 3, 5 or 7 eighths that drift against each other and against the bar, while every cycle start still lands exactly on the grid. It would also make dotted values available, which aren’t possible with the current divisions.

The same option for clock outputs would allow clock dividers like “every 3 eighths” or “every 5 sixteenths.”

Combined with summing LFOs externally, this would allow evolving, non-repeating modulation that stays rhythmically locked to the music.

Thanks for the great module!

xustafu, 2026-10-29:
Added latest request to the list!
We hear you and we are woring on nice things in the back of the shop. Expectg a new testing  firmware soon!

xustafu, 2026-10-29:
> 21\. Hi,
>          feature request: “length” mode for the LFO (and clock) multiplier, for polymetric patterns
>          Currently, the multiplier speeds up the selected division: 1/8 × 3 means three cycles per eighth note. That’s great for polyrhythms, but every multiplier other than a power of two puts the LFO’s cycle off the beat grid.
>          It would be very useful to have an alternative multiplier mode where the value sets the length of a cycle in units of the selected division. For example, with 1/8 selected:
>          ×1 = a cycle of one 8th note
>          ×2 = a cycle of two 8th notes (a quarter note)
>          ×3 = a cycle of three 8th notes (a dotted quarter)
>          ×5 = a cycle of five 8th notes, and so on
>          This would make polymetric patterns possible: LFOs with cycle lengths of 3, 5 or 7 eighths that drift against each other and against the bar, while every cycle start still lands exactly on the grid. It would also make dotted values available, which aren’t possible with the current divisions.
>          The same option for clock outputs would allow clock dividers like “every 3 eighths” or “every 5 sixteenths.”
>          Combined with summing LFOs externally, this would allow evolving, non-repeating modulation that stays rhythmically locked to the music.
>          Thanks for the great module!

@bartofzo 
In theory this could be achieved already with clocks. I changed the test website so it shows differnet labeling. 
https://befaco.org/configmidithing2NEW/
This website only works with latest RC firmware.
Can you check if how the clock behaves is what you are looking for here?
applying it to LFOs might  need soem more work.

> 17\. Hi Befaco team!
>          I’ve noticed is that the Clock outputs only offer PPQN division and multiplication, and all clocks always align to the start of a bar. That means it’s not possible to generate off-beat/upbeat clocks (like eighth-note upbeats between quarter kicks) directly from the module without external logic or delay modules.
>          Could you consider adding an optional clock phase/offset or delay control on the clock outputs? Even something simple like a 50% offset or a delay expressed in musical divisions (e.g. +1/16) would open up a lot of rhythmic possibilities without changing the core identity of the module.
>          Thanks for considering it, and for all the awesome work you do! 🎶

This might be a nice addition indeed :)