Befaco/MIDIThing2 #47: High Noise in the Output

jotel-california, 2024-12-05:
I recently got a hertz donut mk3, and sadly it does not play well at all with the midi thing. The pitch is jumping around a lot. 
When I connect it to any other CV source, it works just fine. Since I was curious, I got my oscilloscope and I saw that there is signifitcant noise on any CV Output. 

It's a constant pulse train with its frequency at 1.25kHz. The amplitude varies with the output, in the lower octaves it is around 300-400mV, the higher you go, the amplitude gets a bit lower, in the higher octaves it's still at 100-200mV

Any idea what could be the problem here? These values are a LOT of fluctuation, no wonder the hertz donut is acting up. 
This is very bad behaviour. Any Idea what might cause this?

I'm using the Konstant Lab Strong Power as PSU and for testing I disconnected everything but the midi thing. 

Made a video as always: https://youtu.be/GrW9jD4Qot0

xustafu, 2024-12-07:
Cannot reproduce this one. Checking in the scope cannot see this noise pattern nor notice any anomality on the tuning
Tested three diffenet units with Pony VCO, Plaits and OXI coral.

Could you try with a differnet power supply? is this noise noticeable with other VCOs?

Anyway, considering you were having such big  tuning issues and now this noise anormality, maybe we are facing a roge unit?

jotel-california, 2024-12-08:
It really seems my unit is defective.
I definitly can say, this a midi thing problem.
I tried it with 3 different PSUs, outside of the case with just the midi thing plugged directly into the PSU.
PLugged into a singular outlet on my wall with nothing else connected.

This would definitly explain my tuning issues. So far I mostly used analog VCOs, and the hertz donut is known to be susceptible to fluctuations in the CV. The other VCOs probably filtered out the inaccuracies, but still averaged the cv, hence the tuning difference. 

Should I contact support for this one? Bought it not even half a year ago, should be covered under warranty.
I did have issues with it when I got it. It bricked itself after trying to save a patch. It was only revived by hard wiping the teensy, just uploading a new FW didnt solve it. Maybe sth got messed up.

xustafu, 2024-12-09:
Do not think that the issues you had re-programming teensy might be causing any of this.
We know that there is some dirt on the CV signal. but not that much dirt as you describe. Also, we have tested with a wide variety of oscillators and all were behaving just fine.
From your comment I can assume the noise is not present in other oscillators you tried right? Did you also tested it in other digital oscillators? I did it with Plaits and Coral succesfuly.
I'm asking because I fear that Hertz Donut is going to be more sensitive to any CV disturbance and even after replacing your unit the module is still going to be affected.
As a test, make a cable or adapter with a 10uF capacitor from tip to ground connector. This should help with Hertz donut.
Anyway, where did you buy the unit? we can try to replace it via the shop or us directly if that makes you more confident. But I fear that Hertz donut is still gonna be a problem!

jotel-california, 2024-12-09:
Just contacted support, I ordered it directly from your site, so that should be fine. 
Sending the Hertz Donut back anyway, but still would like a Midi Thing without the noise.

xustafu, 2024-12-09:
Can you confirm that the noise is only happening with the hertz donut? Did you had noise issues with other oscillators?
We do know that there is some noise in the CV outs (please be warned), but not noticeable in any of the oscillators we tested. We are trully interested to know if this is hertz donut related or wider issue, that we cannot reproduce.
Anyway, we will test your unit once arrives, please follow my colleagues instructions to get it replaced.

jotel-california, 2024-12-09:
Yes, the noisy fluctuation in pitch only happens with the hertz donut mk3.

The other oscillators are just tracking a bit weird (see other threads with special firmware solution)

xustafu, 2024-12-09:
Good.
With tracking weird you mean the tracking problems reported in #43 ?
these issues are alñready solved with the callibration right?

xustafu, 2024-12-09:
Could not reproduce that one either :(
[MidiThing2_v1_2_TEST4.zip](https://github.com/user-attachments/files/18062250/MidiThing2_v1_2_TEST4.zip)
This is the latest version, please give it a go and report in that issue a way to reproduce that behavior.
Will wait for your unit to arrive, test it and then close this issue.

jotel-california, 2024-12-13:
You are right, with your new TEST4 Firwmare, this bug is gone. Nice.
The only minor thing is, that the note name for the 0V point is not displayed. By default 0V is C-2, and if I play this note, the note display shows "-" as if no note was present. The Gate and everything else works, so this is just visual. 

And just so you know, I'm sending you my unitfor diagnosis of the pitch problem.